dchristl/esp32_nat_router_extended #117: Probleme de materiel?

gemmeetik, 2024-04-12:
Bonjour tout le monde, j'ai un esp32 - wroom - 32U mais visiblement en récupérant le .bin pour esp32 cela ne fonctionne pas, j'ai testé avec les autres , rien non plus. Est ce le modèle d'esp qui ne convient pas? (j'ai testé avec plusieurs, rien n'y fait). lors du flash les points s'enchainent et ne s’arrêtent pas (1h d'attente)

Si quelqu'un sait si cela vient du modèle....

Merci d'avance

dchristl, 2024-04-13:
Hello @gemmeetik, 

I'm not entirely sure where and how you're trying to download something (perhaps it's also a translation issue, could you please write in English)? The firmware should be downloaded in the browser and then flashed via cable, which should only take a few seconds. It shouldn't depend on the model.

Kind Regards,
Danny

gemmeetik, 2024-04-15:
@dchristl 
Google
google traduction
Tous
[Images](https://www.google.com/search?client=firefox-b-d&sca_esv=df9114d5be7f4ca8&sca_upv=1&sxsrf=ACQVn0-YuVlznAD8rJb5yEtrTBQbFHvhWQ:1713172882198&q=google+traduction&uds=AMwkrPufgvfzFeIw5yilA4Gzc5SOV5jqiAJsR4Q-4xZxn8kFMCTeE4AuEB3rxbGWhAoxdx9J4EYhBjTSp-rUpb9lNbdg3t7dABClo2WRgtLIJY9ohCa08s-4FYky8mEp6YmXUqVcoGSTldMp9ijvUb_Rkw2E4_NUaRgH6zd61jOYDTfyXIauphcjnwasUoessvYCJZFYIALqg-wIyfonFQUuDawN7hi-eKKwBJPiBOt1aAbvYBq5bPfS--ttYo52GBqIZ2m0My2b&udm=2&prmd=ivbnz&sa=X&ved=2ahUKEwjb7dOo8sOFAxUJdqQEHRlYC7AQtKgLegQIDRAB)
[Vidéos](https://www.google.com/search?client=firefox-b-d&sca_esv=df9114d5be7f4ca8&sca_upv=1&sxsrf=ACQVn0-YuVlznAD8rJb5yEtrTBQbFHvhWQ:1713172882198&q=google+traduction&tbm=vid&source=lnms&prmd=ivbnz&sa=X&ved=2ahUKEwjb7dOo8sOFAxUJdqQEHRlYC7AQ0pQJegQIDBAB)
[Livres](https://www.google.com/search?client=firefox-b-d&sca_esv=df9114d5be7f4ca8&sca_upv=1&sxsrf=ACQVn0-YuVlznAD8rJb5yEtrTBQbFHvhWQ:1713172882198&q=google+traduction&tbm=bks&source=lnms&prmd=ivbnz&sa=X&ved=2ahUKEwjb7dOo8sOFAxUJdqQEHRlYC7AQ0pQJegQIDxAB)
[Actualités](https://www.google.com/search?client=firefox-b-d&sca_esv=df9114d5be7f4ca8&sca_upv=1&sxsrf=ACQVn0-YuVlznAD8rJb5yEtrTBQbFHvhWQ:1713172882198&q=google+traduction&tbm=nws&source=lnms&prmd=ivbnz&sa=X&ved=2ahUKEwjb7dOo8sOFAxUJdqQEHRlYC7AQ0pQJegQIDhAB)
Plus
Outils

    SafeSearch

Environ 531 000 000 résultats (0,28 secondes) 
Français
Anglais

Good morning,
What I explained is that I have several 32-WROOM-32U esps and when I download the bin file (whichever bin file is chosen) I have a download of the file for the esp which takes more half an hour and which always ends with an error. does the problem come from the esp model? but I can't get the program to install correctly on the esp.
Thanks in advance

gemmeetik, 2024-04-15:
@dchristl 
Good morning,
What I explained is that I have several 32-WROOM-32U esps and when I download the bin file (whichever bin file is chosen) I have a download of the file for the esp which takes more half an hour and which always ends with an error. does the problem come from the esp model? but I can't get the program to install correctly on the esp.
Thanks in advance

gemmeetik, 2024-04-15:
[image]
[image]
[image]

ok, so I download flash dowload tool from the github archive, then I download the compressed folder esp32nat-extended_full_v7.0.0.0 and I decompress it, I follow the information from github to flash the esp and impossible to flash it, I have a series of points which never stops (as in the photo) I tried with the different types of bin file and indicating at the start of esp32flash download tool the corresponding esp but that does not change anything in the flashing

gemmeetik, 2024-04-15:
after 22mn:
[image]

and it never stops.....

gemmeetik, 2024-04-15:
and esp is on port 4
[image]
perhaps the bauds on flash download tool and esp ?
[image]

dchristl, 2024-04-15:
It seems like nothing has been flashed or the process didn't work. Typically, when using the ESP Flash Download Tool and starting the flash process, the window (command line) in the background that looks like [this](https://private-user-images.githubusercontent.com/2836933/252157575-48f4937a-f80c-4ff3-96f8-ebfe1fa743dc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMxOTIwNzIsIm5iZiI6MTcxMzE5MTc3MiwicGF0aCI6Ii8yODM2OTMzLzI1MjE1NzU3NS00OGY0OTM3YS1mODBjLTRmZjMtOTZmOC1lYmZlMWZhNzQzZGMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MTVUMTQzNjEyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2MzOTM0YTc2N2FmNmNmOTA5ZThjN2M2YmNmYWU3OTRkOTg1YTUyODk0YjBlOWRmNjczZWRiMjY0MTMwMDk3NSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.MRE92Tbh4GbI4-jNoj2Is08dbBRQLsaeLi6ruYFGO2A)

Could you check if you have the correct com port. Can you do a simple erase? Does this work? Do you have another cable? There are several chips out there, maybe you have some that have to bes set in flash mode, but I'm not sure. Maybe, check your manual. 

Btw. the correct zip for your device is esp32nat_extended_full_v7.0.0.zip

gemmeetik, 2024-04-15:
@dchristl ok it seems it is the cable, but now i've got another bug:
"[2024-04-15 17:06:18,700][ESP8266Loader_spi[1]][espDownloader.py][line:773][ERROR]:
no log file output ...

gemmeetik, 2024-04-15:
@dchristl ok it's good, i try 3cables , and unfortunately it seems that the usb port... it works fine! thks!

closed 2024-04-15 by gemmeetik