dchristl/esp32_nat_router_extended #118: Wifi password not saved

Matthieu33k, 2024-04-15:
Hello,

First of all, thank you for this project. 

I have a problem when I write my Wifi password: after rebooting, esp32 doesn't save the password of my Asus router (placed behind my internet router). On the other hand, it does save the password for the Orange router.

dchristl, 2024-04-15:
Hello @Matthieu33k ,

does the password contain any special characters that might be causing an issue? Could you post the password (or parts of it)? Have you tried temporarily changing the password on the router? What password is it changed to if the saving process fails (blank or something cryptic)?

Matthieu33k, 2024-04-15:
Hi, 

My password is 49 characters long. With letters, numbers and special characters. I haven't changed the router password yet. I'll try tomorrow. 

Thanks for your rapid response !

dchristl, 2024-04-21:
I've found the issue. If the password is too long and has many special characters it doesn't fir in the available storing space no more. I will provide a hotfix release for that .

closed 2024-04-21 by dchristl