dchristl/esp32_nat_router_extended #122: XIAO ESP32-C3 not booting (v7.1.0 & v7.1.1)

Mainak009, 2024-05-22:
Hello, previously I was using this firmware on **ESP32 WROOM DEVKITV1** and it's working fine till the latest update v7.1.1.....but I recently bought the **[XIAO ESP32C3](https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html)** and want to try this because of it's faster CPU and long antenna range. When I flashed the latest v7.1.1 it wasn't booting at all and throwing device descriptor failed on the Windows device manager which is likely a bootloader failure issue....then I tested some previous versions and noticed the same with v7.1.0 too.....The only last working version is v7.0.0 and all the previous ones....please check what is causing the issue and fix it ASAP....I'm willing to participate in any beta test if you want, also let me know if any debug information needed.

Product: XIAO ESP32C3
Product link: https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html
Chip: ESP32-C3
Flash size: 4MB
Flashing address: 0x0

dchristl, 2024-05-24:
Hello,

could you please share a [serial connection log](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/establish-serial-connection.html)? My C3 works fine with any release.

Mainak009, 2024-05-25:
hello...I tried as you said but 7.1.1 isn't even connecting in normal serial mode.
every time this error(check attached screenshot) if I normally connect with those 2 problematic versions, only force download mode to connect in the COM port....but with v7.0.0 it's connecting serial COM normally.... I guess your C3 is a normal one...XIAO is a little different

[image]

dchristl, 2024-05-26:
There is normally no difference between any C3 chip of any vendor, because the chip (the one that is programmed) is always the same. 
The firmware has also no impact on the Windows driver. It seems your setup is faulty, maybe your cable, too. I can't help you, if you don't provide a log, even a flash log. I don't know if Arduino IDE will work (I don't use it), but if your Windows setup is correct it should.

Mainak009, 2024-05-26:
all my PC, USB ports and cables are okay.
FYI normally ESP32 devkit has a separate serial converter but XIAO ESP32C3(and other supermini versions) has a built-in one that requires support from the bootloader to connect in serial mode normally...You need to research more on XIAO ESP32C3 before questioning my setup.

here is the v7.0.0 log which I got normally from Arduino IDE but not in v7.1.0 & v7.1.1 because of the serial connection issue
```
[0;32mI (335) esp_image: segment 4: paddr=00159cc0 vaddr=403860[0;32mI (2648) phy_init: phy_version 1130,b4e4b80,Sep  5 2023,11:09:30[0m
I (2678) wifi:mode : softAP (ec:da:3b:be:75:9d)
I (2678) wifi:Total power save buffer number: 16
I (2678) wifi:Init max length of beacon: 752/752
I (2678) wifi:Init max length of beacon: 752/752
[0;32mI (2698) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1[0m
[0;32mI (2698) ESP32NRE: wifi_init_ap with default finished.[0m
[0;32mI (2708) DNSServer: Socket created[0m
[0;32mI (2708) DNSServer: Socket bound, port 53[0m
[0;32mI (2708) DNSServer: Waiting for data[0m
[0;32mI (2708) DNSServer: DNS Server started[0m
[0;32mI (2718) ESP32NRE: On board LED is enabled[0m
[0;32mI (2728) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (2728) ESP32NRE: NAT is enabled[0m
[0;32mI (2748) ESP32NRE: Starting config web server[0m
[0;32mI (2748) HTTPServer: Keep alive is disabled[0m
[0;32mI (2748) HTTPServer: Starting server on port: '80'[0m
[0;32mI (2758) HTTPServer: Registering URI handlers[0m

ESP32 NAT ROUTER EXTENDED
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

Unconfigured WiFi
Configure using 'set_sta' and 'set_ap' and restart.
```

can't take log using putty as it's not connecting to COM port, is there any other way to take serial log? also don't understand what you mean by taking flash log?

dchristl, 2024-06-02:
> FYI normally ESP32 devkit has a separate serial converter but XIAO ESP32C3(and other supermini versions) has a built-in one that requires support from the bootloader to connect in serial mode normally...You need to research more on XIAO ESP32C3 before questioning my setup.

That doesn't change my statement at all, and it doesn't make any sense. The chip doesn't know what it's built on. The software programs the chip and not any attached peripherals. If you want me to research further (are you aware this is a hobby project?), then please provide a link to the Espressif documentation that supports your statement. The bootloader has never changed since version 1  of the router and doesn't change during updates. Such errors are 99% due to incorrect setup or incorrect flashing. That's why I asked for the flash log. This is the log that appears during flashing. It's the output from esptool or the CMD window in the ESP Flash Download Tool. I want to know how it's being flashed and what the tool says (OK/not OK).

Mainak009, 2024-06-02:
[image]
[image]

Here you can see both v7.0.0 & v7.1.0 are the same and say OK...but v7.1.0 isn't booting or connecting at all to the UART serial when disconnect and connect normally without holding any buttons....which it should as v7.0.0 does.....and I'm saying again my whole PC, cables and everything is okay and fine

dchristl, 2024-06-03:
I see several potential problems:
- the folder contains spaces, this leads often to problems
- the SPI mode is wrong (should be DIO)
- the DonotChgBin checkbox is checked.

There is even a screenshot in the documentation, how to flash properly: 
https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/win_flash_full.png

You can also try:

- Erase the chip before flashing
- Update any older (working) version with OTA-update or by flashing the firmware only (without bootloader and partition) 

This is also described in the documentation.

Mainak009, 2024-06-03:
Yes I forgot to mention....I even tried to update from working v7.0.0 to v7.1.1 which completed successfully but same not booting problem after updating....I also tried other spi modes and folder without space (from desktop) too but that doesn't matter as the tool automatically put "" in the full path

I'll try to erase before flash and update you

kondakos, 2024-06-05:
I had the same issue with XIAO ESP32-C3 as https://github.com/dchristl/esp32_nat_router_extended/issues/119
The only working solution was to use esp32c3nat_extended_full_v6.5.0.bin.

Mainak009, 2024-06-05:
> I had the same issue with XIAO ESP32-C3 as https://github.com/dchristl/esp32_nat_router_extended/issues/119
> The only working solution for mine was to use esp32c3nat_extended_full_v6.5.0.bin.
> 

I've tried last working one is v7.0.0....but dev is so lazy excusing this and that BS🥲😑
Exact same issue as mine device not recognised.... @dchristl how can both our drivers, PC, cable get bad only with v7.1.0 and later and working with v7.0.0?.....I'm telling you in this XIAO there's something not normal than normal ESP32-C3....you need to get one for testing otherwise you never know what is wrong....and as @kondakos said in his reference issue, doesn't matter the v7.1.0(or later) install from OTA or normal flash it's getting the whole chip corrupted somehow.

dchristl, 2024-06-10:
@kondakos 

Your last response in the linked issue was.:
> The fw7.1.1 installed successfully via OTA update.

That clearly means it worked. I'm confused. 

> but dev is so lazy excusing this and that BS🥲😑

Apparently you are right, because you seem to know better how all this works. You've probably already read the documentation in full and now know exactly what to do. I just seem to be talking nonsense. 
But yes, I'm too lazy to reply to spoiled brats who insult me. For a project that I provide in my spare time, free of charge, at the cost of hardware and with considerable hours of work. So, good luck, this was my last answer to you.

closed 2024-06-10 by dchristl