dchristl/esp32_nat_router_extended #124: Connection problems in a mesh WiFi network

SHLHH59, 2024-06-03:
Hello,
my intention is to improve the connectivity of another ESP32 WLED installation in my garage. 
Therefor I'm testing the nat-router since a couple of weeks in my mesh WiFi network, consisting of 3 FritzBoxes in my house.
After power up the nat-router connects properly to the closest FritzBox with approx. -62db.
For some reason unfortunately after some days (when exactly is not known) the nat-router reconnects to another FritzBox in my house.
That particular FritzBox is furthest away from the nat-router and the signal strength is now between -79 and -83 db.
The only way to resolve this and reconnect to the closer FritzBox is to power down the nat-router and power it up again.

The nat-router firmware (7.1.1) is running on a ESP32-WROOM-32UE with an external WiFi-antenna.
Flash size is 8MB.
Any suggestions how to improve this situation and to esteblish a connection to the best uplink signal?

Thanks

SHLHH59, 2024-06-11:
Hi Danny,

many thanks for Your reply.

In a Fritz-Mesh setup ‘normally’ the Mesh-Master determines the settings of all connected Mesh-Clients. This applies, among other settings, to SSID and WLAN pwd.
However, it seems to be possible to disable the automatic adoption of the settings from the Mesh-Master so that settings of the Mesh-Client can be set individually (I never tried that).
The backdraft of doing so: all existing mobile WLAN devices would need to be set to the changed WLAN settings of the relevant Mesh-Client. 
Do You think it would improve the situation if I only set the radio channel individually instead of automatic and leave SSID and pwd as set in the Mesh-Master?  

On the other hand I, have a handful of other ESP’s, running ESPEasy, in my network for which I never encountered this problem so far.

Kind regards
Stefan

[image]
[image]

SHLHH59, 2024-06-11:
After sending my previous response, I have repowered the nat-router several times, changed the position of the device, done a couple of network scans but it never connected again to the strongest radio signal, as it did before.
And this although I did not change any settings in my FritzBox yet.
That’s really strange.

dchristl, 2024-06-17:
That all sounds very strange, unfortunately, I can't really help you with that. On one hand, my mesh knowledge is rather limited, and on the other hand, with the ESP32, there is almost no way to intervene with which router it connects to. You can't do that with other devices either. It seems as if the mesh network itself determines which device connects where.

SHLHH59, 2024-06-17:
Thanks for Your response.
Meanwhile I guess (and hope) to have the problem solved. 
After changing the position of the nat-router several times (even in close distance to the target Mesh-Client) without improvement, I removed the external antenna. After that indeed the connection to related Mesh-Client was successful.
So, finally I've choosen an antenna with less radio amplification and since a couple of days the nat-router stays connected to the stronges WiFi-signal.

Thanks and cheers
Stefan

SHLHH59, 2024-06-20:
Hi,
I'm afraid, I was too optimistic. 
After some days I noticed Yesterday, that the nat-router has again connected do the most far away FritzBox with the weakest signal.
Even after some repower the router does not connect to the strongest WiFi signal again.
So unfortunately I have to give up this project.
Anyhow, many thanks
Stefan

dchristl, 2024-06-22:
I'm sorry to hear that, but I'm afraid there's nothing I can do.