dchristl/esp32_nat_router_extended #129: ESP32-S2 slow transfer

piotr-go, 2024-07-14:
Hi
I have 19Mb/s transfer speed on ESP32 and ESP32-C6 but only 4Mb/s on ESP32-S2.
I suspect not enough RAM may be a problem. Can you enable PSRAM support for S2 ?

Soft: 7.1.1
Board: https://www.wemos.cc/en/latest/s2/s2_mini.html

Thank You

dchristl, 2024-07-29:
Hello,

unfortunately I can't do that. I try to keep the code base as general as possible and ignore specifics for individual devices.   
Sorry, it's not worth it and it's also extremely time consuming to support several devices and configurations when you can simply buy another (fully supported) chip for €3. I hope you understand that.

Kind Regards,
Danny

piotr-go, 2024-07-29:
From what I remember, you can set autodetection of PSRAM in the config.

piotr-go, 2024-08-01:
My ESP32-C6 module: ESP32-C6-WROOM-1-N8
From: https://pl.aliexpress.com/item/1005006271962503.html
I only set ssid and password.

My AP: MikroTik RB952Ui-5ac2nD

I-hate-2FA, 2024-08-01:
source router is wifi6 and can do 100m
my module is ESP32-C6-WROOM-1-N4

I-hate-2FA, 2024-08-01:
do your register as wifi6 ap to client?

piotr-go, 2024-08-01:
Unfortunately my AP is only Wi-Fi 4.
I don't have Wi-Fi 6 AP to test.

piotr-go, 2024-08-01:
Phone also wi-fi 4, but i have wi-fi 6 (RTL8852BE) in my PC.

> sudo iw wlp40s0 link
> Connected to xx:xx:xx:xx:xx:xx (on wlp40s0)
>         SSID: xxx
>         freq: 2452.0
>         RX: 2785081 bytes (6118 packets)
>         TX: 271165 bytes (1652 packets)
>         signal: -29 dBm
>         rx bitrate: 27.0 MBit/s MCS 1 40MHz
>         tx bitrate: 150.0 MBit/s MCS 7 40MHz short GI
>         bss flags: short-preamble short-slot-time
>         dtim period: 2
>         beacon int: 100