dchristl/esp32_nat_router_extended #131: ESP32-C3 does not work with version 7.1.1 and 7.1.0

piotr-go, 2024-07-24:
ESP32-C3 does not work with version 7.1.1 and 7.1.0.
Last working version 7.0.0.
Board: https://pl.aliexpress.com/i/1005005780121305.html

> ESP-ROM:esp32c3-api1-20210207
> Build:Feb  7 2021
> rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fcd5820,len:0x1704
> load:0x403cc710,len:0x968
> load:0x403ce710,len:0x2f9c
> entry 0x403cc710
> I (30) boot: ESP-IDF 5.1.2 2nd stage bootloader
> I (30) boot: compile time Apr 26 2024 15:00:38
> I (30) boot: chip revision: v0.4
> I (33) boot.esp32c3: SPI Speed      : 80MHz
> I (38) boot.esp32c3: SPI Mode       : DIO
> I (43) boot.esp32c3: SPI Flash Size : 4MB
> I (47) boot: Enabling RNG early entropy source...
> I (53) boot: Partition Table:
> I (56) boot: ## Label            Usage          Type ST Offset   Length
> I (64) boot:  0 nvs              WiFi data        01 02 00009000 00005000
> I (71) boot:  1 otadata          OTA data         01 00 0000e000 00002000
> I (79) boot:  2 ota_0            OTA app          00 10 00010000 00177000
> I (86) boot:  3 ota_1            OTA app          00 11 00190000 00177000
> I (93) boot: End of partition table
> I (98) esp_image: segment 0: paddr=00010020 vaddr=3c0e0020 size=671a8h (422312) map
> I (175) esp_image: segment 1: paddr=000771d0 vaddr=3fc92800 size=02e1ch ( 11804) load
> I (177) esp_image: segment 2: paddr=00079ff4 vaddr=40380000 size=06024h ( 24612) load
> I (185) esp_image: segment 3: paddr=00080020 vaddr=42000020 size=d9d18h (892184) map
> I (334) esp_image: segment 4: paddr=00159d40 vaddr=40386024 size=0c660h ( 50784) load
> I (345) esp_image: segment 5: paddr=001663a8 vaddr=50000000 size=00020h (    32) load
> I (351) boot: Loaded app from partition at offset 0x10000
> I (351) boot: Disabling RNG early entropy source...
> I (365) cpu_start: Unicore app
> I (365) cpu_start: Pro cpu up.
> I (374) cpu_start: Pro cpu start user code
> I (374) cpu_start: cpu freq: 160000000 Hz
> I (374) cpu_start: Application information:
> I (377) cpu_start: Project name:     esp32_nat_router_extended
> I (383) cpu_start: App version:      7.1.1
> I (388) cpu_start: Compile time:     Apr 26 2024 15:00:09
> I (394) cpu_start: ELF file SHA256:  33f8c113446b93c4...
> I (400) cpu_start: ESP-IDF:          5.1.2
> I (406) cpu_start: Min chip rev:     v0.3
> I (410) cpu_start: Max chip rev:     v0.99 
> I (415) cpu_start: Chip rev:         v0.4
> I (419) heap_init: Initializing. RAM available for dynamic allocation:
> I (427) heap_init: At 3FC9A3E0 len 00025C20 (151 KiB): DRAM
> I (433) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
> I (440) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
> I (447) heap_init: At 50000020 len 00001FC8 (7 KiB): RTCRAM
> I (455) spi_flash: detected chip: generic
> I (458) spi_flash: flash io: dio
> I (463) sleep: Configure to isolate all GPIO pins in sleep state
> I (469) sleep: Enable automatic switching of GPIO sleep configuration
> I (476) app_start: Starting scheduler on CPU0
> I (481) main_task: Started on CPU0
> I (481) main_task: Calling app_main()
> I (491) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (491) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> W (1501) ESP32NRE: Reset Pin (GPIO 18) set for 1s
> I (1501) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> W (2501) ESP32NRE: Reset Pin (GPIO 18) set for 2s
> I (2501) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> W (3501) ESP32NRE: Reset Pin (GPIO 18) set for 3s
> I (3501) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> W (4501) ESP32NRE: Reset Pin (GPIO 18) set for 4s
> I (4501) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> W (5501) ESP32NRE: Reset Pin (GPIO 18) set for 5s
> W (5501) ESP32NRE: Device will be resetted! Disconnect bridge and reboot device afterwards.
> I (5501) cmd_nvs: Namespace 'esp32_nat' was  erased
> I (5501) main_task: Returned from app_main()

dchristl, 2024-07-29:
Hello, 

I am testing with an ESP32 C3 and generally both versions work. Your output also looks like it works. One output is just strange : 

> W (5501) ESP32NRE: Reset Pin (GPIO 18) set for 5s
W (5501) ESP32NRE: Device will be resetted! Disconnect bridge and reboot device afterwards.
I (5501) cmd_nvs: Namespace 'esp32_nat' was erased

Have you bridged GPIO 18, like desribed [here](https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/advanced.md#resetting-the-device-erasing-the-flash) or is it used otherwise?
It looks as if your device resets itself after every restart.

piotr-go, 2024-07-29:
GPIO18 and GPIO19 are USB pins.
I found some crappy USB cable, no data lines, only power, and it's working :) Thank You.
It appears that powering this module via a regular USB cable is not possible with the current firmware.
With normal USB cable module can't be found over wifi.

dchristl, 2024-08-05:
I' ve changed this recently because of #105  and problems with the default pins for reset. It seems different boards have different pins for USB. I will have a look into this and hope to find a solution, when I have more time.