dchristl/esp32_nat_router_extended #132: Restrict access of websites

akira-kun, 2024-08-06:
How do I restrict the router to open only a small list of websites? Or do a blacklist of forbiden websites...

davidalami, 2024-09-05:
Hey guys!
@dchristl, the repo is awesome, I would like to contribute in the direction mentioned by @akira-kun. 
My idea is to take a blacklist of websites (e.g. [this one](https://someonewhocares.org/hosts/)) in order to block ads. 
I have already tried to do that  by introducing a `bool is_blocked_domain(const char* domain)` function to [DNS server code](https://github.com/dchristl/esp32_nat_router_extended/blob/master/src/dnserver.c). It works correctly (checked by logs), but only until I connect the repeater to actual WIFI access point. After that, I see in the logs that the DNS server is disabled. In the DNS settings GUI I choose AdGuard to make sure that it does not rely on the access point DNS. 
Could you please advise me on how to prevent the application to disable the internal DNS server after connection to access point?

dchristl, 2024-09-06:
Hello,

there is currently no way to filter websites, as the ESP32 is simply too weak to filter websites efficiently. This is why the internal DNS server only runs briefly to redirect all requests to the login page of the router. 

@davidalami 
> Could you please advise me on how to prevent the application disabling the internal DNS server after connection to access point?

The DNS server is stopped when the router receives an IP.  (esp32_nat_router.c:461). If you deactivate this, however, you would have to ensure that all queries that the router cannot handle (e.g. Google) are translated by the upstream DNS.

Kind Regards,
Danny

davidalami, 2024-09-06:
Hello Danny,

Thanks for clarifying, now I get why we're using the internal DNS mainly to redirect to the login page now. If I manage to get some stable working solution, will create a PR.

Cheers,
David