dchristl/esp32_nat_router_extended #133: I need to customize The project

Shubhransu04, 2024-08-07:
In order to develop my own product i am willing to customize it. So i need the full source code and how to convert alll the files like all the html files and all other files into a single .bin file. I am already using this Esp32 nat extended full v 7.1.1 , but i need to customize it can you share it with me I will implememnt this on a real world prduct. I am willing to use atleast 200-300 nos of esp32 on a single project site in a mesh protocol, so in order to do that i need the original code and all the originatl htmal and css file . Please share the details if possible contact me personally i can tell you what i need to customize.
Our clients- Indian Railway, Large no of educational institutions and also a large no of residential appartments and commercial buildings.
i need also suggestions which other hardware can i use for a greater speed of like 200 mbps over the wireless mesh network. Cost doesn't matter but should be customizable, and easy to use, should be implemented in wireless mesh network, there is no chance of wired connection. so I need suggestions for the product. If possible there should be single or multiple ufl ports for external anetnnas for increasing the range of the product to several hundreds of meters.

Slavegrid, 2024-08-07:
"Please sars, I need you to make me money, sars. Your compensation will be knowing that you made me money. Also, I need 10x the max throughput that the best ESP32 can support."

This is why I don't host my open source projects on Github.

Shubhransu04, 2024-08-09:
Can you give me it personally.

closed 2024-08-13 by Shubhransu04