dchristl/esp32_nat_router_extended #140: Failed to connect to ESP32-S3: No serial data received

saimanohar1999, 2024-09-22:
Great work much appreciated.

----->>>>>comming to my issue I was able to flash this project 
https://github.com/martin-ger/esp32_nat_router/blob/master/main/esp32_nat_router.c

---->>>> but your project i am unable to do getting the following error.

administrator@che-ll-caeteam01:~/400k/esp/esp-idf/examples/esp32_nat_router$ idf.py -p /dev/ttyUSB0 -b 115200 flash
Executing action: flash
Running ninja in directory /home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/build
Executing "ninja flash"...
[1/5] cd /home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/bu...ator/400k/esp/esp-idf/examples/esp32_nat_router/build/esp32_nat_router.bin
esp32_nat_router.bin binary size 0xfbb80 bytes. Smallest app partition is 0x12c000 bytes. 0x30480 bytes (16%) free.
[1/1] cd /home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/bu...400k/esp/esp-idf/examples/esp32_nat_router/build/bootloader/bootloader.bin
Bootloader binary size 0x4970 bytes. 0x3690 bytes (43%) free.
[2/3] cd /home/administrator/400k/esp/esp-idf/components/esptool_py && /us...administrator/400k/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32s3 -p /dev/ttyUSB0 -b 115200 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 bootloader/bootloader.bin 0x10000 esp32_nat_router.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.8.dev5
Serial port /dev/ttyUSB0
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
CMake Error at run_serial_tool.cmake:66 (message):
  
  /home/administrator/.espressif/python_env/idf5.3_py3.10_env/bin/python;;/home/administrator/400k/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed.

FAILED: CMakeFiles/flash /home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/build/CMakeFiles/flash 
cd /home/administrator/400k/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/home/administrator/400k/esp/esp-idf -D "SERIAL_TOOL=/home/administrator/.espressif/python_env/idf5.3_py3.10_env/bin/python;;/home/administrator/400k/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/build -P /home/administrator/400k/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/build/log/idf_py_stderr_output_552507 and /home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/build/log/idf_py_stdout_output_552507

saimanohar1999, 2024-09-22:
with platform io also i am unable to flash

Processing esp32-s3 (board: esp32-s3-devkitm-1; framework: espidf; platform: espressif32@6.5.0)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitm-1.html
PLATFORM: Espressif 32 (6.5.0) > Espressif ESP32-S3-DevKitM-1
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50102.240122 (5.1.2) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
Warning! Flash memory size mismatch detected. Expected 8MB, found 2MB!
Please select a proper value in your `sdkconfig.defaults` or via the `menuconfig` target!
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Retrieving maximum program size .pio/build/esp32-s3/firmware.elf
Checking size .pio/build/esp32-s3/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  11.4% (used 37380 bytes from 327680 bytes)
Flash: [========= ]  88.2% (used 1354161 bytes from 1536000 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: /dev/ttyUSB0
Uploading .pio/build/esp32-s3/firmware.bin
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 2
========================================================== [FAILED] Took 11.26 seconds ==========================================================

Environment    Status    Duration
-------------  --------  ------------
esp32-s3       FAILED    00:00:11.255
===================================================== 1 failed, 0 succeeded in 00:00:11.255 =====================================================

 *  The terminal process "platformio 'run', '--target', 'upload', '--environment', 'esp32-s3', '--upload-port', '/dev/ttyUSB0'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

dchristl, 2025-02-05:
Hello,

have you compiled the code for the ESP32-S3 yourself, as I do not provide a release for it? If this is the case, then you must also adjust all the parameters in the config (s [here](https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/advanced.md#modified-parameters-compared-to-the-default-configuration))