dchristl/esp32_nat_router_extended #145: password issue

flylowgofastturnleft, 2024-11-09:
Hello, all work fine but when I set an AP password, I can't connect after the ESP reboot ?

dchristl, 2025-02-05:
Hello,

first check whether it is generally not possible with a password that is as simple as possible. Alternatively, check whether any special characters are copied when copying the password (e.g. password manager).
If none of this works, then check the log or look directly into the variable for the password in the [cli](https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/advanced.md#command-line-interface).

dchristl, 2025-02-05:
I've tested it out. 
- changed only AP password -> working
- changed only STA password -> working
- changed STA and AP password -> working
- changed AP password and SSID-> working

That would be a huge problem if it didn't work. Can you describe the exact workflow? What do you enter where, when and which button is pressed, etc.?