dchristl/esp32_nat_router_extended #153: Future request: WiFi to Ethernet Bridge

dvinesky, 2025-04-04:
It would be nice if this could include a WiFi to Ethernet bridge so that devices like the WT32-ETH01 that has a built-in ethernet port can be used to provide internet for devices that don't have wifi built in.  This could also work with ESP32 device with a LAN8720 device.

NellowTCS, 2025-04-04:
> It would be nice if this could include a WiFi to Ethernet bridge so that devices like the WT32-ETH01 that has a built-in ethernet port can be used to provide internet for devices that don't have wifi built in. This could also work with ESP32 device with a LAN8720 device.

Not advertising or anything, but I am working on something like this, using the WT32-ETH01 or a W5500 Ethernet Module with an ESP32-S2.

However, it's currently not fully finished, and I will be fixing/working on it, so I'll update this when I do that.

dvinesky, 2025-04-11:
> Not advertising or anything, but I am working on something like this, using the WT32-ETH01 or a W5500 Ethernet Module with an ESP32-S2.
>
> However, it's currently not fully finished, and I will be [...]

Awesome.  Let me know if you get it to work.  I have been looking into this as well.