dchristl/esp32_nat_router_extended #157: Problems flashing esp32s3nat_extended_full_v7.2.0 on ESP32-S3 DevKitC-1

Prinsx-py, 2025-05-24:
I’m experiencing problems flashing the prebuilt firmware esp32s3nat_extended_full_v7.2.0.bin on ESP32-S3 DevKitC-1 boards (N16R8). I followed all the instructions exactly, using the recommended settings (40MHz SPI speed, DIO mode, offset 0x0, baud 115200).

Symptoms include:
1. Serial monitor shows various “invalid header” errors
2. Sometimes no serial output
3. Occasionally, the device boots an AP named “ESP32-C6” with password 12345678, which doesn’t match the firmware
4. Configuration UI never appeared when it had flashed and the AP name was "ESP32-6"

I have tried multiple flashing methods (official ESP32 Flash Download Tool, esptool.py on different machines, flashing via phone OTG), and tested with two boards with the same results.

Other ESP32-S3 example firmware flash and run fine on these boards, so the hardware and flashing setup should be okay.

Please advise if there is a known issue or different flashing procedure for this firmware or board.

Thanks!

nouritto, 2025-06-06:
i faced the same problem on my ESP32-S3-Geek and solved it by doing:

download ESP flash download tool v3.9.7
download this firmware ---> https://github.com/dchristl/esp32_nat_router_extended/issues/106#issuecomment-1976491491
first erase the chip then flash the firmware using those settings:
	bootloader.bin	@ 0x0
	partitions.bin	        @ 0x8000
	firmware.bin	        @ 0x10000
	SPI Speed	        40MHz
	SPI Mode	                QIO
	DoNotChgBin    	Yes
then flash the ota update ---> esp32s3nat_extended_v7.2.0.bin @ 0x10000
disconnect the device then connect

it works like magic :)

nouritto, 2025-06-06:
I encountered the same issue on my ESP32-S3-Geek and resolved it with the following steps:

1. download ESP flash download tool v3.9.7
2. download this firmware ---> https://github.com/dchristl/esp32_nat_router_extended/issues/106#issuecomment-1976491491
3. Erase the chip before flashing
4. Flash the firmware using the following configuration:
	bootloader.bin @ 0x0
	partitions.bin @ 0x8000
	firmware.bin @ 0x10000
	SPI Speed: 40MHz
	SPI Mode: QIO
	DoNotChgBin: Yes
5. After that, flash the OTA update ---> esp32s3nat_extended_v7.2.0.bin @ 0x10000
6. Disconnect the device and reconnect it.

it works like magic :)

80G5q1, 2025-07-06:
> I encountered the same issue on my ESP32-S3-Geek and resolved it with the following steps:
> 
>     1. download ESP flash download tool v3.9.7
> 
>     2. download this firmware ---> [Request for ESP S3 chip #106 (comment)](https://github.com/dchristl/esp32_nat_router_extended/issues/106#issuecomment-1976491491)
> 
>     3. Erase the chip before flashing
> 
>     4. Flash the firmware using the following configuration:
>        bootloader.bin @ 0x0
>        partitions.bin @ 0x8000
>        firmware.bin @ 0x10000
>        SPI Speed: 40MHz
>        SPI Mode: QIO
>        DoNotChgBin: Yes
> 
>     5. After that, flash the OTA update ---> esp32s3nat_extended_v7.2.0.bin @ 0x10000
> 
>     6. Disconnect the device and reconnect it.
> 
> 
> it works like magic :)

Also thanks from me. For my ESP32-S3-Wroom1 N16R8 it worked with the following esptool commands

esptool erase_flash
esptool write_flash -ff 40m -fm dio 0x00000 esp32-s3-firmware/bootloader.bin 0x8000 esp32-s3-firmware/partitions.bin 0x10000 esp32-s3-firmware/firmware.bin
esptool write_flash -ff 40m -fm dio 0x10000 esp32-s3-firmware/esp32s3nat_extended_v7.2.0.bin

For me it worked only with SPI mode DIO (-fm dio)